GrumpyOldPizza/arduino-STM32L4 #2: a small wishlist

ramonschepers, 2016-07-26:
since i don't know how the HAL works, i thought i'd say my ideas for now. 

USB:
selectable USB modes (raw hid, flash storage, MIDI instument)

MPU:
MPU support

SPI:
DMA support 
16bit transfers
32bit transfers

Support:
a RTOS

More examples:
low power modes
SPI with DMA example
MPU example

GrumpyOldPizza, 2016-07-26:
USB: Good thought, but that will take time. 

MPU support is not planned. At the end of the day you need a mechanism to statically declare your MPU regions (assigned to IO regions, executables, stack and so on). Doing that dynamically, efficiently and within the resources available is next to impossible. I did implement a uITRON 4.0 RTOS system that could do that, but there are many details that made this unattractive.

SPI: The STM32L4 SPI peripheral has no 32bit wide support, only up to 16bit. DMA is already supported via this API here:

     // STM32L4 EXTENSTION: asynchronous composite transaction
    bool transfer(SPISettings settings, const void *txBuffer, void *rxBuffer, size_t count, void(*callback)(void), bool halfDuplex = false);

RTOS: There will be RTOS support real soon (after SDIO for an SDCARD).

rajdarge, 2016-07-26:
USB host?

GrumpyOldPizza, 2016-07-26:
Sorry, no USB host. There was no space on the PCB to have the necessary additional components. N.b. that there were also no free GPIOs ...

ramonschepers, 2016-07-26:
@GrumpyOldPizza so you said spi can transfer 16 bit data natively, how does that work with the dma? (code example?)

GrumpyOldPizza, 2016-07-26:
@ramonschepers The Arduino spec only implements 8 bit data. If you are using SPI mode 3, it does not make a difference of you have a 8 bit or a 16 bit data stream (other than byte swapping). For a ILI9341 for  example (which wants to have 16 bit color data), that works good enough.

Code example:

   SPI.transfer(settings, &tx_data[0], NULL, 256, NULL);
   while (!SPI.done()) { continue; }

So this send data via SPI, 256 bytes, but does not receive data. The callback is NULL, and it's polling. So really, really bad. Ideally you have a callback that gets called when the transfer is completed. In that callback you can either kick off the next transfer, or update the CS signal.

kriswiner, 2017-01-21:
I think it is a good idea to have  wish list and I expect users will use this space to ask both dumb and profoud questions, both will help us improve the core and the user experience.