GrumpyOldPizza/arduino-STM32L4 #5: error when uploading sketch?

ramonschepers, 2016-11-14:
so i try to compile my stuff, which succeeds, then my Arduino IDE wants to upload it to the board, the blue flashing led goes off, and i get an error message 
"There was an error during upload with uploading the sketch"

however, when i reset my dragonfly (reset switch), it simply runs the initial bootloader blinky again.

my IDE version: Arduino.cc IDE 1.6.11
Latest board package (had the same with version .16 which i had before)
OS:win 8.1 which is up to date (including drivers)

could it be my 'programmer' setting?
i currently tried the st-link, stm32 bootloader, and arduinoisp

GrumpyOldPizza, 2016-11-14:
Can you let it run first till it blinks green ? The blue blinking is the formatting of the SFLASH, which takes a few minutes.

Does Windows show the serial port device properly ?

ramonschepers, 2016-11-14:
the green led is LED on and the blue LED is blinking, plus once i plug the board in, i get the device in 'my computer' as should.

the Serial port is there, i tried it on a usb 2.0 and usb 3.0 port, same issues

i also tried cwapping drivers with Zadig, no success

ramonschepers, 2016-11-14:
update: i have let it blink for half an hour now, and realized this was the first sketch that i got to work a period ago, so it's normal the blue led flashes.

but now it fails somehow...

also, the ide gets the dragonfly in bootloader mode (no leds on), when in bootloader mode, i can only see the device in device manager (device status says this: device won't start)

GrumpyOldPizza, 2016-11-14:
Can you use Zadig when the device is in bootloader mode to associate it with "winusb" ? Perhaps Win 8.1 decided to install another driver there ?

I'd switch the device manually into bootloader mode by press/release the RESET button while holding the BOOT button.

closed 2016-11-21 by GrumpyOldPizza

ramonschepers, 2016-11-21:
still the same error in arduino, though the device is recognized as "STM32_BOOTLOADER" in the device manager

GrumpyOldPizza, 2016-11-21:
Did you use Zadig to put winusb on the STM32 BOOTLOADER device?

ramonschepers, 2016-11-22:
yes, also, going to try another pc now, checking if it's a software config issue

ramonschepers, 2016-11-22:
i can confirm it is a config issue on my pc, as it works on my second laptop, but where else to look for..

GrumpyOldPizza, 2016-11-22:
I have no idea what the issue is with your PC. Perhaps having 2 devices with the same VID/PID might be an issue.

ramonschepers, 2016-11-22:
don't know, i do know it's working on my older laptop, which is good enough for me :)