GrumpyOldPizza/arduino-STM32L4 #7: USB CDC on Nucleo 476RG

TTrml, 2017-02-24:
Hi,

Thanks for this awesome core! It makes it so much easier to program those STM32L4 chips...

I am trying to add a CDC UART on a Nucleo 476 board, with a micro USB adapter on pins PA9/PA11/PA12. This setup is programmed as a Dragonfly variant, via the onboard ST-Link.

When plugged through its USB CDC port, the board is correctly recognized by Windows, and I can connect to it with a terminal application on the computer. It receives the strings sent via the serial terminal on the PC, but an exception is raised on the microcontroller when it tries to send some data to the PC, and the data never makes it to the terminal application.

Any clue as to what might be going on? Is the USB CDC code only compatible with the Dragonfly's L476RE, and not the Nucleo's L476RG chip?
Thanks,

Armel

GrumpyOldPizza, 2017-02-24:
That actually should work. You should be able to do that without PA9, by
using this define in variant.h:

#define STM32L4_CONFIG_USB_VBUS           GPIO_PIN_NONE

I'd not recommend using the "Dragonfly" entry for that. Just add #define
USBCON to variant.h (and the upper define), and modify boards.txt to
include the USB settings for the Nucleo board. "Dragonfly" uses different
clocking internally.

Mind trying the simple example called "Asciitable", so that we are on the
same page ?

- Thomas

TTrml, 2017-02-24:
Thanks Thomas for the quick response!

I followed your advice and got the "Asciitable" example to work... but surprisingly enough, if I take away the content of the loop() function, then the part printed in setup() doesn't show up, even though it is displayed when the content of loop() is kept!

Here is the code that doesn't print anything, even though the "Asciitable" example does:
-
void setup() {
	//Initialize serial and wait for port to open:
	Serial.begin(9600);
	while (!Serial) {
		; // wait for serial port to connect. Needed for native USB port only
	}
	// prints title with ending line break
	Serial.println("ASCII Table ~ Character Map");
}
void loop() {
}
-

GrumpyOldPizza, 2017-02-24:
Excellent feedback. Is this with the latest code off github ? Or a
downloaded release via json file ?

- Thomas

TTrml, 2017-02-24:
The code is from Github as of 3 days ago, and the same happens with the current code.

TTrml, 2017-02-24:
Ah just got it now!
Text is only displayed if it exceeds a certain length - for instance, the previous example works if "ASCII Table ~ Character Map" is replaced with "ASCII Table ~ Charapppppppppppppppppppppppppppppppppppppppppcter Map", in which case the whole lot is shown

GrumpyOldPizza, 2017-02-24:
Yup. There is a mechanism to avoid sending too small USB/CDC packets. I
changed the code yesterday to use USB SOF callbacks over the previously
SysTick based approach. So something is not flushing the packets. Thanx for
investigating.

- Thomas

GrumpyOldPizza, 2017-02-25:
Fixed. STM32L433 always turned on SOF, STM32L476 needed a flag for that.

Is the crash still an issue ?

TTrml, 2017-02-25:
All good now, thanks a lot Thomas!

closed 2017-02-25 by TTrml