GrumpyOldPizza/arduino-STM32L4 #11: Some questions

edogaldo, 2017-04-21:
Which is the reference library you rely on? SPL? Cube? A custom one?
What's the purpose of "boot_stm32l4xx.c"?

GrumpyOldPizza, 2017-04-21:
(1) Custom system level code. No SPL, no Cube other than USB/PCD support

(2) OTA (IAP), preboot checking (gateway to USB/DFU loader, RTC setup).

edogaldo, 2017-04-21:
Thank you!
Another question: I was investigating the Reset_Handler and I don't understand the branches targets, i.e.:
        b.n     3f
        ...
        bne.n   1b
        ...
        b.n     6f
        etc...

What are 1b, 3f, 6f etc.?! they are not labels, are they?!

GrumpyOldPizza, 2017-04-21:
Which file are you referring to ? boot_stm32l4xx is the preboot code. It
that succeeds it branches to the rest reset code which is located somewhere
else.

edogaldo, 2017-04-21:
Sorry, no, now I'm referring to startup_stm32l476xx.S

GrumpyOldPizza, 2017-04-21:
Ok, those are just normal assembly labels ... 6f == 6th label forward ...

- Thomas

edogaldo, 2017-04-21:
Thank you! I didn't know that notation..

> Sorry, no, now I'm referring to startup_stm32l476xx.S
>
> > Which file are you referring to ? boot_stm32l4xx is the preboot code. It
> > that succeeds it branches to the rest reset code which is located
> somewhere
> > else.
> >
> >
> > > Thank you!
> > > Another question: I was investigating the Reset_Handler and I don't
> > > understand the branches targets, i.e.:
> > > b.n 3f
> > > ...
> > > bne.n 1b
> > > ...
> > > b.n 6f
> > > etc...
> > >
> > > What are 1b, 3f, 6f etc.?! they are not labels, are they?!
> > >
> > > <https://github.com/GrumpyOldPizza/arduino-STM32L4/issues/11#
> > issuecomment-296252411>,
> > EXzHLxTw4DywFrRxBvks5ryOY7gaJpZM4NEi89>
> > >
> >
> > <https://github.com/GrumpyOldPizza/arduino-STM32L4/issues/11#
> issuecomment-296262847>,

> ATTZbh4RdPBbNHETXu74eMEjLkEHd3bWks5ryPBmgaJpZM4NEi89>

> <https://github.com/GrumpyOldPizza/arduino-STM32L4/issues/11#
issuecomment-296265706>,
AG4QfHz5uXvcZxoqBXruSGtBiyz8bSNxks5ryPMpgaJpZM4NEi89>

closed 2017-04-28 by GrumpyOldPizza