GrumpyOldPizza/arduino-STM32L4 #12: compiling wonderfull in eclipse arduino plugin, but code recognition ist broken

mindforger, 2017-04-28:
I added the json to my locations and loaded a new sketch using the boardfiles for my NUCLEO-L432KC
I can compile and modify it without issues and it runs, but the whole project won't recognize its includes, all includes are "unresolved" and macros and types are undefined

did i just miss something? haven't had this in eclipse projects ... last time I remember was when I acidentially used another compiler but the project setting was still in "built-in" mode and had not details on how to serve the project ... i can not find any discrepancies like that yet as far as i have seen

mindforger, 2017-04-28:
fixed it, your board.txt under the nucleo l432 section is missing usb_manufacturer, usb_product and usb_type
also i removed the dosfs section in the platform.txt because it yould not be build because of missing entrys in the board.txt ... but i honestly have no clue what to enter there :D

GrumpyOldPizza, 2017-04-28:
I am not familiar with the plugin you are using. But the missing defines
and such are all in boards.txt / platform.txt. So you seem to have a
different problem with the setup.

- Thomas

mindforger, 2017-04-28:
i can not comply
https://github.com/GrumpyOldPizza/arduino-STM32L4/blob/master/boards.txt#L198
VS
https://github.com/GrumpyOldPizza/arduino-STM32L4/blob/master/boards.txt#L279

in the NUCLEO-L423 Section usb-manufacturer and usb-product are missing ... i replaced them with some arbitrary names ... also the dosfs entrys are missing, after i removed the dosfs section from platform everything worked out just well!
[boards.txt](https://github.com/GrumpyOldPizza/arduino-STM32L4/files/964295/boards.txt)
[platform.txt](https://github.com/GrumpyOldPizza/arduino-STM32L4/files/964294/platform.txt)

those work wonderfully now

GrumpyOldPizza, 2017-04-28:
> txt#L198
> txt#L279

NUCLEO-L432 builds just fine with the Arduino IDE. So there seems to be a
problem with the boards.txt / platform.txt on your end.

- Thomas

closed 2017-04-28 by GrumpyOldPizza

mindforger, 2017-04-28:
i never said it does not build, i only said the files are incomplete and the arduino parser can not build the dependencies to provide code completion and can not resolve datatypes, i found the reason in an earlier issue somewhere else with the hint that all "used" macros need to be present or the IDE can not resolve the includes to provide code completion ... it would only mean to inset those lines i showed you in my files and maybe fill the dosfs variables with some dummy values ... it would be an improvment

GrumpyOldPizza, 2017-04-28:
Not really. This is all handled in boards.txt either via selectable
options. That is done on purpose.

NUCLEO-L432 does not support USB or DOSFS to begin with. In that case
nothing is defined, and hence there should not be a compile problem to
begin with. So I suspect you are hitting a build problem with Eclipse ...

To be more exact, USBCON is defined in the relevant "variant.h" file. If a
variant (like NUCLEO-L432) has no USB, then it's not defined. The code in
USBCore.cpp is guarded by "#if defined(USBCON)". Same thing is true for the
various DOSFS_* defines which normally come in via menu options in the IDE.
If none is defined, then DOSFS is not reference.

So I am utterly confused what the problem at hand is.

- Thomas