GrumpyOldPizza/arduino-STM32L4 #14: PIN_MAP missing

Jorgen-VikingGod, 2017-06-07:
is there a way to get similar information like in other STM32 boards by the PIN_MAP?
like this one:
https://github.com/arduino-org/arduino-core-stm32f4/blob/e14d9abe4c33e3e37db11218227b7683333c5837/variants/otto/variant.cpp

I want to bring the NeoPixelBus library to life, but it uses something like this to send out the pixel values:
```
...
uint32_t  pinMask = BIT(PIN_MAP[pin].gpio_bit);
volatile uint16_t* set = &(PIN_MAP[pin].gpio_device->regs->BSRRL);
volatile uint16_t* clr = &(PIN_MAP[pin].gpio_device->regs->BSRRH);
...
```

GrumpyOldPizza, 2017-06-07:
> e14d9abe4c33e3e37db11218227b7683333c5837/variants/otto/variant.cpp

There are the standard macros, which OTTO seems to be missing:

#define digitalPinToPort(P)        ( g_APinDescription[P].GPIO )
#define digitalPinToBitMask(P)     ( g_APinDescription[P].bit )
#define portInputRegister(port)    ( (volatile uint32_t*)((volatile
uint8_t*)(port) + 0x10) ) // IDR
#define portOutputRegister(port)   ( (volatile uint32_t*)((volatile
uint8_t*)(port) + 0x14) ) // ODR
#define portSetRegister(port)      ( (volatile uint32_t*)((volatile
uint8_t*)(port) + 0x18) ) // BSRR
#define portClearRegister(port)    ( (volatile uint32_t*)((volatile
uint8_t*)(port) + 0x28) ) // BRR
#define digitalPinHasPWM(P)        ( g_APinDescription[P].attr &
PIN_ATTR_PWM )

AVR has those, SAM has those and SAMD has those. Unfortunately there is no
standard type for them. For STM32L4 it's a void*

- Thomas

Jorgen-VikingGod, 2017-06-07:
thx Thomas. I changed the code above by this and it is working now:
```
        uint32_t pinMask = g_APinDescription[pin].bit;
        GPIO_TypeDef *GPIO = g_APinDescription[pin].GPIO;

        volatile uint16_t* set = &(GPIO->BRR);
        volatile uint16_t* clr = &(GPIO->BSRR);
```

closed 2017-06-07 by Jorgen-VikingGod

GrumpyOldPizza, 2017-06-07:
You might want to use "volatile uint32_t" for set/clr.

- Thomas

Jorgen-VikingGod, 2017-06-08:
good point - I will change this too into the NeoPixel lib. Thank you for your support