GrumpyOldPizza/arduino-STM32L4 #15: Butterfly board not recognized by automatic board selection

seamaster101, 2017-07-29:
Hi Thomas,
I'm playing with Butterfly board from Kris Winer. I had success compiling and uploading code from the Kris' examples.  
This Issue surfaced later, when I added some more code to his example. This code ran fine on Teensy3.2 before I tried to migrate to Butterfly.
I contacted with the author of the library that I was using and he pointed me out in the direction here:
************
"I think automatic board selection does not recognize your board. You can force it to use avr by defining on the beginning
#define USE_N2K_CAN 4

before your line
#include "NMEA2000_CAN.h"
************
I did that, but then I got different error
*********** 
"n file included from /Volumes/MrBig/Users/Jordans/Documents/Arduino/libraries/NMEA2000-master/NMEA2000_CAN.h:88:0,
                 from /Volumes/MrBig/Users/Jordans/Documents/Arduino/Ultimate_sensor_fusion_compass_V0.2/Ultimate_sensor_fusion_compass_V0.2.ino:52:
/Volumes/MrBig/Users/Jordans/Documents/Arduino/libraries/avr_can-master/avr_can.h:78:10: error: #error Unsupported CPU in avr_can.h
         #error Unsupported CPU in avr_can.h
          ^
/Volumes/MrBig/Users/Jordans/Documents/Arduino/libraries/avr_can-master/avr_can.h:218:10: error: #error CPU Frequency F_CPU value not supported in avr_can.h
         #error CPU Frequency F_CPU value not supported in avr_can.h"
**********
Obviously that (AVR) wasn't the right CPU architecture.
My question is if there a compatibility issue with STM32L processor and Arduino sketches, or is something in my install not correct? 
I did compile the code examples successfully, so I'm thinking it is not the install. Is there plan to make the transition to these new boards easier? I did transition from Arduino to Teensy3.2 and that was really smooth with out any major issues. I like the Butterfly board, but I had few issues trying to transition.
Any direction you can provide?
thanks,
Jordan

GrumpyOldPizza, 2017-07-29:
I'd say it's as simple as the NMEA2000 library not being properly configured. If you send it down the AVR path, it tries to use a AVR specific hardware setup that of course is not gonna work with STM32L4. 

Assuming you have an external MCP2515 (or similar shield), you might try "USE_N2K_CAN 1". 

N.b. I am just guessing there, as you did not provide any links or such for me to find out what you are really using. The NMEA2000 source I found on github is rather hardware specific, so it's unlikely that anything will work by chance, other than their supported setups. STM32L4 is not one of them.

closed 2017-08-14 by GrumpyOldPizza

seamaster101, 2017-08-16:
Hi,
Thanks for the replay. Here the link to the library I've been using. 
Apperantly I need low level CAN driver for the Butterfly board that for Teensy was provided by Paul. 
I was wondering if you guys plan to write CAN driver for the STM32L4, so people like me can utilize the functionality of the board?
 I'm not that good to write my own:(

seamaster101, 2017-08-16:
Oops forgot the link https://github.com/ttlappalainen/NMEA2000?files=1

GrumpyOldPizza, 2017-08-16:
There might be a CAN driver, but I have no test case to validate it. That's
kind of where things get stuck. There does not seem to be any
off-the-shelf-sensor or something where we could try out whether the
functionality is properly implemented.

GrumpyOldPizza, 2017-08-16:
This NMEA2000 library relies upon custom CAN drivers, like
https://github.com/ttlappalainen/NMEA2000_teensy and
https://github.com/ttlappalainen/NMEA2000_mcp.

Hence us doing a proper CAN driver will not help, as this is not what this
library does expect. Seems the best thing to do is to contact the author of
the NMEA2000 library and associated CAN drivers to ask ask for STM32L4
support.

- Thomas