GrumpyOldPizza/arduino-STM32L4 #16: STM32L476 Nucleo USER button doesn't function

woxinwuchen, 2017-08-12:
Hi , I am using STM32L476 Nucleo board. But when I load example code "button". The USER button doesn't take effect. I have test the read back status of USER button via Serial port . It turn out to be random values.
 I have tested example code "blink" which works good. 
And I have tried to upload ST offical demo code hex "STM32CubeL4_Demo_STM32L476RG_Nucleo.hex". It works fine. And the button works too. 
So I suppose it is configuration problem, right? But I have searched in google and other platform . I couldn't find answer. Could you help me?

Here is my configuration and platform.
OS  : tested in MAC OS  Sierra and Win 7 ultimate x64
Arduino IDE 1.8.3
Board : Nucleo-64
Board part number : Nucleo L476RG
Upload method : Stlink
Usb inter face : none
Other serial instance : all
Port : serial port 1423

Code : 
int buttonPin = 2;     // the number of the pushbutton pin

// variables will change:
int buttonState = 0;         // variable for reading the pushbutton status

void setup() {
  // put your setup code here, to run once:
  pinMode(LED_BUILTIN, OUTPUT);
  pinMode(buttonPin, INPUT);
  Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
   buttonState = digitalRead(buttonPin);

  // check if the pushbutton is pressed.
  // if it is, the buttonState is HIGH:
  if (buttonState == HIGH) {
    // turn LED on:
    digitalWrite(LED_BUILTIN, HIGH);
  } else {
    // turn LED off:
    digitalWrite(LED_BUILTIN, LOW); 
  }
    Serial.println(buttonState);

  delay(1000);        // delay in between reads for stability
}

kriswiner, 2017-08-12:
Button is on either pin 39 or 44.

woxinwuchen, 2017-08-13:
@kriswiner Thank you for your reply.  I have just tested with pin 39/44. It still doesn't work.

GrumpyOldPizza, 2017-08-13:
It seems that for the Nucleo the button is D22.

However I am puzzled as to why you don't just use the standard defines for
the user button (which is consistent for all Arduino platforms):

// Other pins
#define PIN_BUTTON           (22)
static const uint8_t BUTTON = PIN_BUTTON;

woxinwuchen, 2017-09-02:
Thank you guys, for your kind help . I have found the reason. Due to I am a newbie. I have added a wrong "Additional Board Manager URL". The URL I added before is from other project. Thank you again!

closed 2018-06-10 by GrumpyOldPizza