GrumpyOldPizza/arduino-STM32L4 #17: Timer interrupts?

laterrps, 2017-08-21:
does exist any way for using timer interrrupts?

GrumpyOldPizza, 2017-08-21:
Not as of right now.

What are you trying to do ?

- Thomas

laterrps, 2017-08-21:
i would like to supervise certain tasks not using if(millis()%x==0) i would like to use timer interrupts.

laterrps, 2017-08-21:
I have been watchin your code and i find out that you are using timers interrupts for dac converters and pwm , is it  too dificult to implement just timmer interrupst?

GrumpyOldPizza, 2017-08-21:
My question was really more like what granularity you need.

The core itself uses internally 1ms timeouts, which you could use.

A Class API does not exist for that yet.

armv7m_timer_t myTimer;

armv7m_timer_create(&myTimer, (armv7m_timer_callback_t)myTimerCallback);

armv7m_timer_start(&myTimer, timeout);

armv7m_timer_stop(&myTimer);

N.b. those timers are one-shot, meaning you need to start the timer again
in "myTimerCallback". Ah, "timeout" is in milliseconds.

Hope that helps.

- Thomas

kriswiner, 2017-08-21:
Or you could use RTC.Alarm but this would be for a 1 Hz or lower frequency
timer.

laterrps, 2017-08-21:
Thank you very much i'll see it! 
best regards!!

laterrps, 2017-08-21:
Thank you very much!!! it works!!! 

armv7m_timer_t myTimer;

armv7m_timer_create(&myTimer, (armv7m_timer_callback_t)myTimerCallback);

armv7m_timer_start(&myTimer, timeout);

armv7m_timer_stop(&myTimer);

closed 2017-12-11 by GrumpyOldPizza