GrumpyOldPizza/arduino-STM32L4 #18: Problems with i2c

laterrps, 2017-08-23:
I have problem using i2c comunication
i am using mpu6050 i dont know why but the program is freezing and interrupts still working

laterrps, 2017-08-23:
this was happening with max30102 sensor too this has been happening at randomly way.

kriswiner, 2017-08-23:
Do you have 4K7 pullups on I2C?

laterrps, 2017-08-23:
[image]

i am using 2.2k resistors

kriswiner, 2017-08-23:
What does an I2C scan show? Can you discover the MPU6050 on the I2C bus.

I have successfully received data from dozens of MPU9250 using the STM32L4
so the problem is likely connection or software.

How are you initializing the wire library?

laterrps, 2017-08-23:
I initialize like this:
 Wire.setClock(reloj);
 Wire.begin();

void movimiento() {
	static bool calculaFC;
	
	if (movread == true) {
		//Serial.println("acc1");
		flagMovimiento=Acel.getIntMotionStatus();
		movimientoActualizado = false;
		movread == false;
		if (flagMovimiento == true && movimientoActualizado == false) {
			/*Serial.println("update1");*/
			tiempoUltimoMovimiento = millis();
			movimientoActualizado = true;
			/*Serial.println("update2");*/

		}
		armv7m_timer_create(&timerMuestreoMovimiento, (armv7m_timer_callback_t)muestraMovimiento);

		armv7m_timer_start(&timerMuestreoMovimiento,200);
		//armv7m_timer_start(&timerMuestreoMovimiento, 100);
		movread = false;
		//Serial.println(flagMovimiento);
	}
	int tiempoActual = millis();
	if (((tiempoActual - tiempoUltimoMovimiento) > 5000) && calculaFC==false) {
		calculaFC = true;
		correct = 0;
		configurarAfe4400();
		//Serial.println("timer");
		armv7m_timer_create(&timerPersistenciaFrecuenciaCardiaca, (armv7m_timer_callback_t)cancelarLecturaFrecuenciaCardiaca);

		armv7m_timer_start(&timerPersistenciaFrecuenciaCardiaca, 40000);
		stopedByTimer = false;
	}
	if (calculaFC) {
		tiempoUltimoMovimiento = millis();
		//Serial.println("dsp");
		DSP(false);
		if (correct == 6) {
			char ppm[10];
			calculaFC = false;
			float fc = auxFrec / 5;
			powerDownAfe4400();
			correct = 0;
			Oled.setDrawColor(0);
			Oled.drawBox(0, 16, 131, 16);
			Oled.setDrawColor(1);
			Oled.drawStr(0,16,"PPM");
			sprintf(ppm, "%.0f", fc);
			Oled.drawStr(50, 16, ppm);
			
		}
		else {
			if (stopedByTimer == true) {
				Serial.println("Error en lectura");
				///////////////AQUI PUEDE IR SINTETIZADOR DE VOZ//////////////////
				////////////////O REINTENTAR CALCULO DE FRECUENCIA CARDIACA///////////
				calculaFC = false;
				powerDownAfe4400();
				Oled.setDrawColor(0);
				Oled.drawBox(0, 16, 131, 16);
				Oled.setDrawColor(1);
				Oled.drawStr(0, 16, "Err.");
				Oled.sendBuffer();
				
				correct = 0;
				stopedByTimer = false;
			}

		}

	}
}

laterrps, 2017-08-23:
I am using mpu library   (Jeff Rowberg)

kriswiner, 2017-08-23:
"I am using mpu library (Jeff Rowberg)"

Yup, that's the problem...

laterrps, 2017-08-23:
or do you have any example?

kriswiner, 2017-08-23:
I havent used the MPU6050 for years having graduated to the MPU9250 but
here is what used to work for me. You will have to modify the wire calls or
maybe not. But there are bound to be some changes required to sun on the
STM32L4.

https://github.com/kriswiner/MPU6050

laterrps, 2017-08-24:
I implemented a motion detect algorithm and it looks like it's working correctly thank you for your help i'll be carefull and instead of copy work from internet i will implement my own code.

kriswiner, 2017-08-24:
Good for you!

laterrps, 2017-08-24:
I'm still having problems i did implement this:

```
Serial.print("status ");Serial.print(Wire.status());Serial.println();
		Serial.print("vector interrupciones  ");Serial.print(interrupcones,BIN);Serial.println();
```

and this shows in terminal.
[image]
and communication get down! but program still running