GrumpyOldPizza/arduino-STM32L4 #19: How can i flash the ladybug stm32

laterrps, 2017-10-06:


kriswiner, 2017-10-06:
Use the Arduino IDE?

laterrps, 2017-10-06:
I would like to use Keil uvision

kriswiner, 2017-10-06:
OK, go ahead...

laterrps, 2017-10-06:
I meant i have user stm32f103 and i programmed it with an ftdi rs232 i just changed boot0 to high and boot1 to low , can I do something similar with your board?

kriswiner, 2017-10-06:
Maybe. If you don;t want to use the Arduino IDE maybe better to use an
ST-Link.

laterrps, 2017-10-06:
ok thank you very much! i will get one stlink! 
best regards

laterrps, 2017-10-06:
I was usind arduino IDE but some times it is necessary to take a look deeper on registers values! so i have gone for keil uvision it have very interesting tools for debugging, I know that some times it is different or dificult when we are starting to use a new tool chain but algo some times it ir neccesari

GrumpyOldPizza, 2017-10-06:
My workflow is to use the "Arduino IDE" to compile, and then "openocd" (which is part of the packge) along with gdb to debug.

closed 2017-10-06 by GrumpyOldPizza