GrumpyOldPizza/arduino-STM32L4 #20: does ladybug have read protection?

laterrps, 2017-10-12:
i have been trying to flash it and i cant

kriswiner, 2017-10-12:
Little more information required.

Are you using the Arduino IDE?

Did you put it into boot mode first (press and hold boot, press and release
restart buttons)?

laterrps, 2017-10-12:
I am using keil uvision and stlink v2

GrumpyOldPizza, 2017-10-12:
Read protection is not enabled. I flash STM32L4 all the time using ST-Link V2. Never used Keil uVision, so perhaps it's just a setup question on your end.

N.b. the debugger cannot break in (and flashing) when the chip is in a low power mode. So pressing toggeling RESET while BOOT is pressed is a good way to get the chip into a good state.

laterrps, 2017-10-12:
[image]

laterrps, 2017-10-12:
[image]

laterrps, 2017-10-12:
can you help me?

laterrps, 2017-10-12:
:/

GrumpyOldPizza, 2017-10-12:
I have zero idea. Does the Arduino IDE work (which is what this github here is about) ?

laterrps, 2017-10-12:
can you help me please?

laterrps, 2017-10-12:
yes it is working!  on arduino ide

GrumpyOldPizza, 2017-10-12:
Ok, so it's either an issue with the ST_Link adapter or the program you are using. Have you tried to update the firmware on the ST-Link (let's assume you have a cheap clone from ebay ...) ? Have you tried to flash this with the official ST ST-Link software to eliminate a Keil uVision problem ?

If it would be read out protected, the normally dfu-utils would throw an error, and the Arduino IDE would fail to flash a sketch.

laterrps, 2017-10-12:
OK ok!!! I dont understand the reason because i can not to flash it!

laterrps, 2017-10-13:
I have programmed now! but it is necessary to conect the rst pin in rst button, plus swdio swclk gnd and 3v3

laterrps, 2017-10-13:
or how do you program it with st link?

laterrps, 2017-10-13:
can you help me with a schematic?

kriswiner, 2017-10-13:
There is a schematic on the Tindie page.

laterrps, 2017-10-13:
i meant the schematic showing me how you programm it

GrumpyOldPizza, 2017-10-13:
There should be a configuration option to use the internal reset via SWD.

kriswiner, 2017-10-13:
Didn't you just say you did program it?

Can you program the Nucleo board or the discovery board?

Maybe the ST forum would be a better place. This site is for the Arduino
core.

GrumpyOldPizza, 2017-10-13:
I am using SWDIO, SWCLK, GND and VCC.

laterrps, 2017-10-13:
maybe is the programmer that i am using!! thank you very much!!! seriously!! you help me a lot!

GrumpyOldPizza, 2017-10-13:
Well, we decided against exposing a RESET pin. Mainly because you can press
RESET manually, and for debugging a live system, the internal reset that
the debugger can trigger via SWD is good enough anyway.

laterrps, 2017-10-13:
@GrumpyOldPizza  i have selecter some diferent options in keil uvision but only this work and i have to "make contact" with a jumper on the button reset pin 
[image]

laterrps, 2017-10-13:
@kriswiner  thank you i talked to you  because i consider you experts! and tank you very mucho for attend my questions! seriousle 
i am sorry if i am disturbing your work!!
best regards!

laterrps, 2017-10-13:
@kriswiner  yes i have not problem with nucleo board now i am working with ladybug ! but i programmed it now, thank you very much

clemaitre58, 2017-10-18:
@GrumpyOldPizza  which version of file do you send by ST-Link 32 V2 *.hex file or *.elf?

closed 2017-10-24 by GrumpyOldPizza