GrumpyOldPizza/arduino-STM32L4 #21: After SPI.end() it's not possible to re-enable SPI with SPI.begin()

millerresearch, 2017-10-14:
Short version: I think stm32l4_spi_disable() should set spi->state to SPI_STATE_INIT instead of SPI_STATE_NONE.

Long version: I've built a [package](https://www.github.com/millerresearch/arduino-mystorm) which extends your arduino-STM32L4 platform to support the [myStorm BlackIce](https://nanode0000.wordpress.com/2017/08/16/getting-started-with-mystorm-blackice/), an open source board combining a STM32L433 with a Lattice ICE40 FPGA.  Thanks to your wonderful work which has done all the heavy lifting for STM32L4, the myStorm package only needs to add a couple of board variant files, and a small library to assist with access to the FPGA.

On the BlackIce, the SPI pins used to configure circuit bitstreams into the FPGA are also multiplexed for other uses (_eg_ driving a set of on-board LEDs). Therefore it would be useful to be able to call SPI.end(), reconfigure the SPI pins for GPIO, and then later call SPI.begin() to reclaim the pins for SPI use again.  Unfortunately this doesn't work because stm32l4_spi_disable doesn't leave spi->state with the value stm32l4_spi_enable expects.  With the change I've suggested above, it does work ... and this is consistent with some of the other drivers (_eg_ i2c, uart) where the xxx_disable() routine leaves the device in INIT state.

GrumpyOldPizza, 2017-10-15:
Sorry for the brief response. The fix for that got checked in last week.

millerresearch, 2017-10-15:
Thanks!  I see now that I should check the github version to keep up to date, and not just rely on the Board Manager.

closed 2017-10-15 by millerresearch