GrumpyOldPizza/arduino-STM32L4 #22: Using PA15 of LadyBug

clemaitre58, 2017-10-24:
I don't know if the question is for @kriswiner or @GrumpyOldPizza .

I would use the STM32L4 core to compile some code for a card which is really close to the Ladybug(L432KC microproc, dfu…). The main difference is that PA15 is route on a (nRF51)component, so I have to use it as an outpout.

My question is how or where make a change in variant.h to allow to use the PA15 pin in a sketch?

Thanks for your help.

Cédric

GrumpyOldPizza, 2017-10-24:
Cedric,

not sure this is possible that easily. PA15 for Ladybug is used to detect
USB_VBUS, which in turn is used to detect a USB connect event (and power
up/down the USB peripheral). What GPIO do you have connected to USB_VBUS ?

- Thomas

clemaitre58, 2017-10-24:
@GrumpyOldPizza Ok I think I understand the idea. Shift my VBUS in the place of PA15, right?

GrumpyOldPizza, 2017-10-24:
Depending upon how you set it up. Create a new variant (boards.txt) and
setup a new variant. In "variant.h" you need to define your USB_VBUS GPIO,
and in "variant.cpp" you'd simply replace the now USB_VBUS pin by PA15.

Mind asking why you did move those pins around ?

clemaitre58, 2017-10-24:
In fact, I don't design the card.
The guy how make this plug PA15 on nRF51. I transmit data above Serial2 when I send of Data. I place a pulse on PA15.

---

An other thing, I think you can add in the readme that the 32-bit support should activated for linux : sudo dpkg --add-architecture i386

If it's not the case you have some crach during compilation when you call the dfu-suffix bin.

Thanks for the help.

clemaitre58, 2017-10-24:
@GrumpyOldPizza a last question : There in the core an eeprom library.

Does this could be used for emulate eeprom in flash memory. In this manner you  write on the eeprom and after a reboot you could access to the data by reading the eeprom again?

GrumpyOldPizza, 2017-10-24:
The EEPROM code for STM32L4 takes a chunk of regular FLASH and emulates an
EEPROM ontop. So yes, after a reboot the data is still there.

clemaitre58, 2017-10-25:
@GrumpyOldPizza is there an example of use?

GrumpyOldPizza, 2017-10-25:
Yes. The stanard examples come with it, documentation is on arduino.cc.

clemaitre58, 2017-11-10:
@GrumpyOldPizza concerning the eeprom : 
when you read the address 0 of the eeprom with address of the flash is it?

TY
Cédric

GrumpyOldPizza, 2017-11-10:
EEPROM emulation is complex. There is no fixed mapping. What you have is a
bunch of dates reserved in regular FLASH that keeps track of what values
should be where in the virtual EEPROM.

- Thomas

clemaitre58, 2017-11-13:
@GrumpyOldPizza where could I find docs or code which explained how is made this none-fixed mapping?

GrumpyOldPizza, 2017-11-13:
https://github.com/GrumpyOldPizza/arduino-STM32L4/blob/master/cores/stm32l4/avr/eeprom.c

clemaitre58, 2017-11-15:
@GrumpyOldPizza Another stupid question : does your core configuration assume that the ladybug have an external oscilator? or you could use the internal oscillator of the STM32L432KC?

TY
Cédric

GrumpyOldPizza, 2017-11-15:
32768Hz is required fir the software stack as is.

- Thomas

clemaitre58, 2017-11-15:
@GrumpyOldPizza   32768Hz external oscillator, rigth?

GrumpyOldPizza, 2017-11-15:
Yes. It's used for RTC and to lock MSI, which drives either SYSCLK
directly, or throu the PLL. Otherwise you need to finetune MSI (and HSI16),
which the current code does not do.

Why would you want to save on 32768Hz ?

The problem not having 32768Hz is that you cannot use LSE. You'd need to
use LSI. And that is frequency wise a random generator. So you needed to
measure it's frequency, with HSI16 as time base. Latter one is only tuned
to about 1% precision. So you'd create overall a system where time is
within +/- 2%.

- Thomas

clemaitre58, 2017-11-15:
@GrumpyOldPizza 

Ok, I understand this. Is there a possibility to use MSI and HSI16 ?
Is it here : https://github.com/GrumpyOldPizza/arduino-STM32L4/blob/master/system/CMSIS/Device/ST/STM32L4xx/Source/system_stm32l4xx.c 

In SystemCoreClockUpdate function?

GrumpyOldPizza, 2017-11-15:
> STM32L4/blob/master/system/CMSIS/Device/ST/STM32L4xx/
> Source/system_stm32l4xx.c

MSI is used for SYSCLK (locked by LSE). HSI16 is used for USART and I2C. So
I am not sure what your question is.

The SystemCoreClockUpdate function really just looks at the RCC values and
from there tries to guess "SystemCoreClock". In any case this file is not
in use for the code in question.

- Thomas

clemaitre58, 2017-11-15:
@GrumpyOldPizza 

So if there is no External oscillator nothing on PC-14 ard PC-15. Does the software in the flash start properly?
Does the component use a default configuration Internal clock?

GrumpyOldPizza, 2017-11-15:
Software starts with  4MHz MSI. Pretty well documented in the datasheet and
the reference manual. From there the code in stages goes to HSI16 and then
later on to MSI+LSE.

- Thomas

clemaitre58, 2017-11-15:
So a binary for Ladybug (which haved LSE) from your core should rum
perfectly (excepted the précision of the clock) on my proto which doesn't
have external oscilator, rigth?

GrumpyOldPizza, 2017-11-15:
No. This STM32L4 Arduino Core REQUIRES LSE (32768Hz) to be connected and
working.

closed 2017-11-19 by GrumpyOldPizza