GrumpyOldPizza/arduino-STM32L4 #23: dfu-suffix

flybrianfly, 2017-11-15:
Writing software for Ladybug and I get this error:
`Build options changed, rebuilding all
fork/exec /home/brian/.arduino15/packages/grumpyoldpizza/hardware/stm32l4/0.0.26/tools/linux//dfu-suffix: no such file or directory
Error compiling for board Ladybug-L432KC.`

I'm using Arduino 1.8.3 on Fedora 27.  Ladybug shows up correctly under "Port" when it's plugged in.

GrumpyOldPizza, 2017-11-15:
Is the dfu-suffix binary in the directory ? Should be there, but perhaps a
download issue when you installed the package ?

It's also a 32bit binary. Does Fedora 27 install the 32 bit libs by default
?

flybrianfly, 2017-11-15:
The dfu-prefix, dfu-suffix, and dfu-util binaries are all there and executable.

Unfortunately, my understanding is that by default Fedora does not install 32 bit libs. Which libs should be installed? Thanks!

GrumpyOldPizza, 2017-11-15:
Mind trying quickly:

sudo dnf install glibc.i686

- Thomas

GrumpyOldPizza, 2017-11-15:
> That fixed the issue. Code builds and uploads correctly now, thanks!

Oki. Thanx for the heads up.

closed 2017-11-19 by GrumpyOldPizza

arthurv, 2018-07-21:
Just commenting for ubuntu-based distros (Linux Mint etc) the command to fix this problem is:

sudo apt-get install libc6-i386