GrumpyOldPizza/arduino-STM32L4 #24: USB board on Dragonfly

arpeggio1234, 2017-11-19:
I am trying to compile an Arduino usb hostshield code and the compiler says ''please define board in avrpins.h''.
How can i do this, i am on Linux, manjaro, and found the avrpins.h file and see there is no 
STM32F4 entry, can i put it in, and yes what? Or can i use the teensy pinout in the code(how does one do that) for the dragonfly since they are lookalikes.
Is this the good place for this question or is there a better place to find answers for this kind of things. I am learning a lot of things especialy looking for answers and where to ask.
Thanks in advance, Dian from the Netherlands

GrumpyOldPizza, 2017-11-19:
Dragonfly has it's own USB port. So why would you want to compile code for
it for a shield ?

Dragonfly also is STM32L4, not STM32F4.

arpeggio1234, 2017-11-19:
Yeah you'r right it's the stm32L4,
the reason for usb is for usb in the to board, do you mean it can except usb in.
On my arduino's i am interfacing with this usb shield to a ps4 controller, very nice and then 
controlling the speak rate and bend(kind of filtering) of a speakjet chip.
So can i attach the ps4 controller direct to the Dragonfly board you think.
Any library names for usb in to stm32L4 board or sites to get more info on that issue.
Thanks for the reply and concern, sir.
Dian

GrumpyOldPizza, 2017-11-19:
Ah, no. USB Host is not supported software wise with Dragonfly.

arpeggio1234, 2017-11-19:
So do you have an idea how to implement the usb host lib to the stm31l4 board.
Include it in the avrpins.h.  do you knw, maybe how to overcome this issue.
Thanks again, Dian

https://github.com/GrumpyOldPizza/arduino-STM32L4/issues/24#issuecomment-345527533

GrumpyOldPizza, 2017-11-19:
> issuecomment-345527533

No, I have no idea. You probably have to port the libray in question and
pick the appropriate pins.

closed 2017-11-19 by GrumpyOldPizza