GrumpyOldPizza/arduino-STM32L4 #25: How to use Serial1  or Serial2 on STM32 L476RG board using GrumpyOldPizza arduino-STM32L4 core

mayukhIITD, 2017-12-10:
Hi,

I need to use NEO- 6M GPS with my STM32 L476RG. I tried to use TinyGPS++.h but it requires SoftwareSerial.h which does not work on 80 MHz boards. Hence I have to use Serial1 as mentioned in:
http://stm32duino.com/viewtopic.php?t=1959
 But the problem is the code that is posted in the above link works for STM32DUINO core (https://github.com/stm32duino/Arduino_Core_STM32) but not on the GrumpyOldPizza core. In GrumpyOldPizza core, Serial1 does not work. Please help me out and let me know how to make Serial1 work on GrumpyOldPizza core??

Thanks,
Mayukh,
IIT Delhi, India

kriswiner, 2017-12-10:
Why doesn't Serial1 work, it works for me using an STM32L476RE and this
core? Maybe you need to adjust the variant file for your board; it might
not use the same pins for Serial1 as this
<https://www.tindie.com/products/TleraCorp/dragonfly-stm32l47696-development-board/>
one.

GrumpyOldPizza, 2017-12-11:
The STM32L476RG NUCLEO board only support "Serial" on D0/D1. That is what
the Arduino UNO standard that the Nucleo follows requires, that's what we
support on this board.

What does not work with "SoftwareSerial.h" ? I have never tested that, but
it implies it's purely some GPIO bitbanging.

N.b. that all of our boards (https://www.tindie.com/stores/TleraCorp/)
support tons of UARTs, as we created proper pinmaps that would allow to do
so.

- Thomas

mayukhIITD, 2017-12-11:
@kriswiner : Thank you so much for your hint !! I had to change the variant files corresponding to STM32L476RG and it worked..
@GrumpyOldPizza : The "SoftwareSerial.h" shows the error:  This version of SoftwareSerial supports only 20, 16 and 8MHz processors when I run it on l476RG board...Anyway now I don't need SoftwareSerial, because I can use Serial1 by modifiying the variant files..

Thanks to both of you for your replies..

GrumpyOldPizza, 2017-12-11:
Kris, should we include a ported version of "SoftwareSerial" ? Looks like
it's just a problem with some delay clocks ... AVR specific ... Not sure
about the value there.

Mayukh, I would recommend against adding Serial1 ... The NUCLEO is really a
bad platform for low-power.

mayukhIITD, 2017-12-11:
Thomas, can you please elaborate why you are recommending against adding Serial1 ? Sorry for this, I am very new to this area. In that case what will be the alternate solution for using GPS on my Nucleo L476RG board?

Thanks,
Mayukh

GrumpyOldPizza, 2017-12-11:
The NUCLEO board is speced with one serial on D0/D1. This is the level that
will be maintained. Any changes you make are not supported.

- Thomas

mayukhIITD, 2017-12-11:
In that case, to add NEO- 6M GPS only SoftwareSerial is what we can rely on?

Thanks,
Mayukh

GrumpyOldPizza, 2017-12-11:
Connect the NEO-6M to D0/D1.

Please also check the official documentation, UM1724, Table 23. That is
what is supported. I do see only one USART2 there, which is mapped to
"Serial".

closed 2017-12-12 by GrumpyOldPizza