GrumpyOldPizza/arduino-STM32L4 #26: can't compile on arm64 arch

ramonschepers, 2017-12-13:
```
fork/exec /home/pine64/.arduino15/packages/grumpyoldpizza/hardware/stm32l4/0.0.26/tools/linux//dfu-suffix: exec format error
Error compiling for board Dragonfly-L476RE.
```
using Arduino 1.8.3 with the package version 0.26, on ubuntu xenial (Release ubuntu 16.04.3 LTS (Xenial Xerus) 64-bit with kernel 3.10)

any idea for a solution?

GrumpyOldPizza, 2017-12-13:
Install the 32 bit libc.

ramonschepers, 2017-12-15:
Seems I can't install it (trying install build-essential:armhf after adding compability for 32bit arm)
Getting errors that say I need a newer g++ which I can't install somehow.

Any suggestion would be to move to arm64 arch sometime, as 32bit  is getting phased out due to it's life span

closed 2018-06-10 by GrumpyOldPizza