GrumpyOldPizza/arduino-STM32L4 #27: Library for X_NUCLEO_IDB05A1

mayukhIITD, 2017-12-22:
@GrumpyOldPizza: Dear Thomas, I want to use X_NUCLEO_IDB05A1 Bluetooth shield with Nucleo L476RG board. If I use STM32DUINO core then the sample code at : https://github.com/stm32duino/X-NUCLEO-IDB05A1 works fine. But when I use it with grumpyoldpizza core then it gives following error:
/home/mayukh/Arduino/libraries/arduino_792491/src/ble_clock.h:24:25: fatal error: wiring_time.h: No such file or directory

Is there any suitable library or example code for using X_NUCLEO_IDB05A1 with grumpyoldpizza core?

Thanks,
Mayukh

mayukhIITD, 2017-12-24:
@kriswiner : Dear Kris, any help on this?

Thanks,
Mayukh

GrumpyOldPizza, 2017-12-24:
wiring_time.h is not a standard Arduino include file. Perhaps you could
replace it by including the official Arduino.h file.

- Thomas