GrumpyOldPizza/arduino-STM32L4 #29: Problem with external RTC on STM32L476RG board

mayukhIITD, 2018-01-03:
Dear Thomas/Kris,

@GrumpyOldPizza @kriswiner
As you know STM32L476RG board has RTC but it is not connected to a Battery. Hence I would like to use an external RTC like ds1307. But when I use DS1307 (I2C address= 0x68) , the usual Arduino libraries like RTCLib  are giving output as:

2165/165/165 (Tuesday) 165:165:85
 since midnight 1/1/1970 = 2275406289s = 26335d
 now + 7d + 30s: 2042/2/15 5:48:15

The same code is running properly when I connect the DS1307 to Arduino board.
Any solution for this will be of much help...

Thanks,
Mayukh

GrumpyOldPizza, 2018-01-03:
RTC on STM32L476RG is connected to a supercap, which keeps RTC running for
about 8 hours. That may or may not be good enough.

With RTCLib, have you checked whether the DS1307 was detected properly and
initialized properly ? With 2275406289s since 1/1/1970, the years 2042
seems to be correct.

mayukhIITD, 2018-01-03:
Thanks for your reply Thomas.  If  the supercap gives 8 hours backup then why is my RTC time resetting everytime the power goes off even for 1 second?

I have checked using I2CScanner that the I2C Adresss of DS1307 i.e. 0x68 is NOT detected..

What is the possible issue then?

Thanks,
Mayukh

GrumpyOldPizza, 2018-01-03:
> Thanks for your reply Thomas. If the supercap gives 8 hours backup then
why is my RTC time resetting everytime the power goes off even for 1 second?

Do you have the variant with L476 or L496 ?

> I have checked using I2CScanner that the I2C Adresss of DS1307 i.e. 0x68
is NOT detected..

Connected to the wrong pins ... no pullups ... wired the wrong way in
general ... wrong I2C speed ...

- Thomas

mayukhIITD, 2018-01-03:
I have L476RG.. Does it have supercap?

Thanks,
Mayukh

GrumpyOldPizza, 2018-01-03:
Yes it does.

- Thomas

kriswiner, 2018-01-03:
There is L476RE or L496RG, the latter has no supercap but a PTH for RTC_BAT
at the top of the board.