GrumpyOldPizza/arduino-STM32L4 #34: STM32F476RE - USB not recognized

denisn73, 2018-04-04:
I'm want use ST25R3911B-DISCO board with this core.
Programm downloading and normal working. But USB dsn't work, it not recognized, it no usb enumiration in windows.
I'm try comment compiling with this and unsuccessfully:
 #define STM32L4_CONFIG_USB_VBUS           GPIO_PIN_NONE
I'm try directly connect 5V to PA9, not happening too.
What can be wrong?

GrumpyOldPizza, 2018-04-04:
Have you tried:

#define STM32L4_CONFIG_HSECLK             0

The board you cite is not supported, so not surprising it ain't working ;-)

Which variant did you start with ?

- Thomas

denisn73, 2018-04-04:
No, but I'll try.
On the board STM32L476RE, respectively, for compilation i use variant "Dragonfly"
USB circuitry differs only in the absence of VBUS and ID lines

denisn73, 2018-04-05:
Tried this settings and nothing has changed:
> #define STM32L4_CONFIG_LSECLK             0 // 32768
> #define STM32L4_CONFIG_HSECLK             0 // 16000000
> #define STM32L4_CONFIG_SYSOPT             0 // SYSTEM_OPTION_VBAT_CHARGING
> #define STM32L4_CONFIG_USB_VBUS           GPIO_PIN_NONE // GPIO_PIN_PA9

denisn73, 2018-04-05:
Added a pull-up resistor to the USB_DP line as well as on boards with STM32F103 and it worked! =))

closed 2018-04-05 by denisn73

GrumpyOldPizza, 2018-04-05:
You need to keep STM32L4_CONFIG_LSECLK 32768

denisn73, 2018-04-05:
I'm known, already keep it and creat separate board variant with pin names

GrumpyOldPizza, 2018-04-05:
Right, but there is no external pullup required. If so, something else is
badly wrong. STM32L476 has a builtin pullup.