GrumpyOldPizza/arduino-STM32L4 #35: I2C issue

tcsaba101, 2018-05-26:
Dear Kris & Thomas,
I have a limited experience in Atmel AVR, ESP32, chipkit PIC development, mainly Arduino, some MPXlab, Atmelstudio + Atmel-ICE platforms. I like Arduino, because of simplicity.

I am kickstarting in STM32 on your platform.
Blink was working on Ladybug, Butterfly and Dragonfly, The Cricket and SensorTile not yet arrived.
Now starting my Sensor-node development experiment, I run in  the first barrier:

[image]

I2c not working.
I can't find the core library like with ESP32.
Anyway I am not expert in this kind of issues.

Please give my some directions to enter stm32l4 arduino world.

Thanks: Csaba Toth
.

kriswiner, 2018-05-26:
This is very old syntax. Check the Ladybug or Dragonfly github repositories
for more current examples.

Wire.begin(TWI_PINS_20_21); // set master mode,
Wire.setClock(400000);             //  I2C frequency at 400 kHz

tcsaba101, 2018-05-28:
Thanks Kris!
Every step looks painful:
This is the example code from Adafruit, what compiles without issue on Arduino Mega.
Please recommend some 1.3" Oled library what works.
I looks that compile as it would be SPI control, but no parameter given, as it is an i2c.
[image]
Where the STM32L4 core stays during compilation? It is temporarily generated during arduino setup?
Does it works together with the STM32L0 core?

GrumpyOldPizza, 2018-05-28:
The use of "portOutputRegister()" and such is sadly utterly importable. AVR
uses 8 bit ports, everybody else uses 32 bit ones.

I am not sure where you got your library from, but the original from
Adafruit has the proper defines in place, that would uses 32 bit ports for
all non-AVR platforms (more or less):

#if defined(__SAM3X8E__)
 typedef volatile RwReg PortReg;
 typedef uint32_t PortMask;
 #define HAVE_PORTREG
#elif defined(ARDUINO_ARCH_SAMD)
// not supported
#elif defined(ESP8266) || defined(ESP32) || defined(ARDUINO_STM32_FEATHER)
|| defined(__arc__)
  typedef volatile uint32_t PortReg;
  typedef uint32_t PortMask;
#elif defined(__AVR__)
  typedef volatile uint8_t PortReg;
  typedef uint8_t PortMask;
  #define HAVE_PORTREG
#else
  // chances are its 32 bit so assume that
  typedef volatile uint32_t PortReg;
  typedef uint32_t PortMask;
#endif

tcsaba101, 2018-05-29:
Thomas you are absolutely right, I have plugged the Adafruit lib from the original source and everything compiled with no issue.

Then it has turned out my Oled from China is not Adafruit compatible. My previous "Adafruit library" was hacked and was handling the fake Oleds, but were not compile on stm32l4 platform. What is compile that is not handling fake parts. Not even worked on Arduino with original library.
So I moved to u8g2 library, what is handling many chips.

Found an example with 2 Oled displays, what was running on Arduino well.
When changed the platform to STM32L4, I have got a new compiling error:
[image]

dtostrf() is not working here or some other mistake happened?

Thanks, Csaba

GrumpyOldPizza, 2018-05-29:
dtostrf() is not part of the Arduino API (nor Standard C Library API).

It's actually part of the Arduino AVR compatibility and you have to use:

#include <avr/dtostrf.h>

That's the same with the official Arduino Cores ...

https://forum.arduino.cc/index.php?topic=390276.0
http://forum.arduino.cc/index.php?topic=368720.0
https://forum.arduino.cc/index.php?topic=349764.0

N.b the STM32L0 / STM32L4 cores use code derived from SAMD. Hence it's the
printf() based dtostrf() variant, which seems to have some issues.