GrumpyOldPizza/arduino-STM32L4 #37: COM port allocation

tcsaba101, 2018-08-21:
Dear Thomas, Kris,

I have a Dragonfly, a Butterfly and a Ladybug boards.
I have uploaded a sketch to a Dragonfly, while the board manager was set to Butterfly and then uploaded a sketch to Butterfly, when board was set to Dragonfly.
Since then I can not setup them to have a COM port on Win10. I have reinstalled the core, the drivers with Zadig as it is written in the ReadMe.

When I press trhe RESET+BOOT buttons, Win10 shows it has found something, bu no COM port appears.
I can't get a COM port allocated to the boards in the Device Manager.
The STM32 BOOTLOADER exist in the Universal Serial Bus Devices category in Win10 Device manager.

[image]

Ladybug is working but it has a plain com port, no USB Device is appearing the same time.:

[image]

What happened? How can I fix it?

Thanks, Csaba

GrumpyOldPizza, 2018-08-21:
Toggle RESET while BOOT is pressed, then simply upload any sketch (ignoring
the COM port allocation).

tcsaba101, 2018-08-21:
It works!

Life is easy when you know how to do it.

I had the blink sketches what I know was worked for me, I used them, after upload the boards got the COM allocation.

Thanks!

Question:
At every reset the Arduino IDE Serial Monitor disconnects (also Putty), and I have to reopen the terminal.  
Is there a solution to have a permanat terminal connection?

kriswiner, 2018-08-21:
Don't think so. I have asked for this "feature" but as far as I know this
is not possible with the ST USB peripheral.

kriswiner, 2018-08-21:
> Not sure what the problem is. When I uploaded a sketch to the board while
> the serial monitor is active, it reconnects.
>
> When I use minicom same thing ...

Not with Windows10 apparently.

tcsaba101, 2018-08-22:
My development background is on Win10 machine. Using the Arduino IDE 328P, ESP32, ESP8266 works well from this point of view.
Depends on terminal application they not loosing the connection during upload or when I start the IDEs Serial Monitor, resets the firmware and I see everything from start.
All those boards, has separate USB/UART converters.

Using STML4 boards, it is not resetting the fw launching the Serial Monitor.
If I press the RESET button by hand, it is loosing the connection when pressing, and reconnect when I release, so you have to restart the terminal by hand immediately after reset.
One solution to leave enough delay at the beginning of the sketch while the USB connects.
Maybe this can be solved on Win7, or this is absolutely relies on the ST embedded USB features?

Anyway there is a buffer (I don't know where Win10/ST?) what keeps from the beginning the sent data, so when terminal starts you see what was printed.

tcsaba101, 2018-08-22:
Additional COM question:
How can I create a COM port for a Butterfly on an older Win machine?
I have created with Butterfly a simple UART > RS485 converter to test a RS485 > SDI12 converter.
In my lab I am using a Vista notebook for HP-IB bus, downloading handheld data, browsing datasheets, etc. Works perfect in this environment.
I tried to connect to a terminal on Vista, but were not able to find a driver for this.
Zadig could not install any ( I used the old 2,2_xp version).

ST-USB can't be use on Vista or XP?

GrumpyOldPizza, 2018-08-22:
As per readme:

USB Serial driver setup for Tlera Corp boards (Window XP / Windows 7 only)
Go to
~/AppData/Local/Arduino15/packages/grumpypoldpizza/hardware/stm32l4/<VERSION>/drivers/windows
Right-click on dpinst_x86.exe (32 bit Windows) or dpinst_amd64.exe (64 bit
Windows) and select Run as administrator
Click on Install this driver software anyway at the Windows Security popup
as the driver is unsigned

This should work on XP / Win7. Vista and Win8 are not supported, however
the might work if you bypass the signature requirement.

tcsaba101, 2018-08-22:
Thanks to light up...
I will install Arduino, then it should work with any terminal application.

Today I have solved, debug has been done.

Next time I will do this way.

Thanks!

tcsaba101, 2018-08-26:
No surprise, it works. As it is written in readme.
Thanks.