GrumpyOldPizza/arduino-STM32L4 #38: conflicting return type specified for virtual int Client::read(uint8_t *buf, size_t size)

dberenguer, 2018-08-27:
Client inherits from Stream but read is declared in Client.h as:

```C++
virtual int read(uint8_t *buf, size_t size) = 0;
```

whilst the parent declaration in Stream.h is:

```C++
virtual size_t read(uint8_t *buffer, size_t size);
```

Using the Adafruit_MQTT library generates the above error. Changing from int to size_t as the return type in Client.h fixes the problem.

GrumpyOldPizza, 2018-08-27:
I'll take a peek. Seems that other libraries (like Wire.h) do use "size_t".
The code in "Stream.h" is an extension from the standard Arduino code.

Looking at "write()" the return type is "size_t".

- Thomas

dberenguer, 2018-08-27:
Perfect. I created a pull request, in case you want to merge.