GrumpyOldPizza/arduino-STM32L4 #40: getVBUS function output

rmeldo, 2018-10-27:
Hi,
A question about the function getVBUS.

I am using a Grasshopper board (LoRa), powered with two 1.5 batteries which give 3.2V. I am powering it through the LDO voltage regulator.

I would like to include the voltage level into the periodic upload message to be aware of the state of charge of the batteries.  

When I call the function "Serial.print(STM32L0.getVBUS());" I get a "1" on the Arduino serial monitor.
Is this what I am supposed to get? How would the message change as the voltage drops?

Thanks
Riccardo

rmeldo, 2018-10-27:
OK, but where does the 3v3 reference voltage comes from, when the battery voltage is 3.2V?

It maybe that I do not understand how analogRead works.
Riccardo

kriswiner, 2018-10-27:
Well, quite right, if you supply the board with 3.2 V the board voltage
will be ~3.1 V or so, so this will be the analog voltage too. This can be
read by STM32L0.getVDDA() IIRC. So you could just read the VDDA and maybe
get a measure of your input voltage, but it won't be exact since for some
reason you are going through the LDO. I wouldn't. Just connect 3.2 V to the
3V3 on the board edge, then the board voltage will be a good measure of
your battery voltage and you can read this via VDDA.

rmeldo, 2018-10-27:
OK so  for a test I am powering through the USB connector and the STM32L0.getVDDA() returns 3.93.
Is it Volts? is seems high. I thought the board ran on 3.3V.
Riccardo

rmeldo, 2018-10-27:
By the way, the intention is to use supply through the 3.3V pin, for the but since I am using a couple of different battery packs I thought it would be safer to go through the LDO

kriswiner, 2018-10-27:
" STM32L0.getVDDA() returns 3.93."

Then something is very wrong with your board...What does 3V3 show?

rmeldo, 2018-10-27:
the 3V3 shows 3.9 as well.   I tried another board and it shows 3.3V. 

could I have damaged the voltage regulator?

kriswiner, 2018-10-27:
Hard to imagine 3.2 V damaging anything, 6 V or 9 V, likely...

rmeldo, 2018-10-28:
Thanks for the advice. STM32L0.getVDDA() is what I was looking for. Still chasing this voltage problem on my board, but that's on me.

Riccardo