GrumpyOldPizza/arduino-STM32L4 #42: Maximum I2C Clock

flybrianfly, 2019-02-07:
What's the maximum I2C clock rate supported by these boards? I see that the boards support up to 1 Mbit/s, but am wondering if the software supports that.

Thanks!
Brian

kriswiner, 2019-02-07:
We have run reliably at 1 MHz bus speed when the sensors can also support
it. I think the L4 will even allow 3.4 MHz bus speeds but few sensors
support this.

GrumpyOldPizza, 2019-02-07:
100kHz, 400kHz and 1000kHz are supported. There is an errata regarding setup time of SDA, which may make 1000kHz fragile in some case. We do use a 16MHz I2CCLK in all cases.

N.b. that those clocks map to "STANDARD MODE", "FAST MODE" and "FAST MODE PLUS", which have slightly different timing requirements.

Please also note that the real speed depends upon your signal paths, pullup resistors and the device(s) on the other side.  So your 400kHz might in real live be closer to 350kHz.

flybrianfly, 2019-02-07:
Thanks! I'm looking for a method transfer data between two STM32L4 microcontrollers and I2C FAST MODE PLUS seems like an easy approach to try.

closed 2019-02-07 by flybrianfly