GrumpyOldPizza/arduino-STM32L4 #43: USB on PA11, PA12

larjohn, 2019-03-04:
Hi there,

thanks for all your hard work. Your library is the only way for me to use I2C on my Radino L4 board (http://wiki.in-circuit.de/index.php5?title=radinoL4_DW1000#Downloads)

The only thing I haven't managed yet, is connect to USB COM port. I can upload my sketches through DFU and they do seem to work, but I have no COM port appear in Windows device manager.

I think butterfly does not use PA11 and PA12 for USB communication, so it might be logical for it to not work. Is there any way to support the COM port for this board as well?

GrumpyOldPizza, 2019-03-04:
Butterfly does us PA11/PA12 for USB. So all of that should work.

larjohn, 2019-03-04:
So I must be doing sth wrong. Can you reproduce the following with butterfly?

0. I connect my board with USB
1. I launch Arduino IDE
2. I select the Butterfly board
3. I open the AnalogReadSerial example (see below)
4. I press both BSL and RST buttons on my board and then let the RST and then the BSL - two leds (RX, TX) indicate we are in the mode where we can upload
5. I upload the example code with the following output:
```
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 25312

Download	[                         ]   0%            0 bytes
Download	[==                       ]   8%         2048 bytes
Download	[====                     ]  16%         4096 bytes
Download	[======                   ]  24%         6144 bytes
Download	[========                 ]  32%         8192 bytes
Download	[==========               ]  40%        10240 bytes
Download	[============             ]  48%        12288 bytes
Download	[==============           ]  56%        14336 bytes
Download	[================         ]  64%        16384 bytes
Download	[==================       ]  72%        18432 bytes
Download	[====================     ]  80%        20480 bytes
Download	[======================   ]  89%        22528 bytes
Download	[======================== ]  97%        24576 bytes
Download	[=========================] 100%        25312 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
```
6. I press the RST button once again. 

I expect that now there should be a COM port in the Windows device manager. All I want to do is debug my code with text printed through the serial port. Am I missing a step?

This is the example:

```
/*
  AnalogReadSerial

  Reads an analog input on pin 0, prints the result to the Serial Monitor.
  Graphical representation is available using Serial Plotter (Tools > Serial Plotter menu).
  Attach the center pin of a potentiometer to pin A0, and the outside pins to +5V and ground.

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/AnalogReadSerial
*/

// the setup routine runs once when you press reset:
void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
}

// the loop routine runs over and over again forever:
void loop() {
  // read the input on analog pin 0:
  int sensorValue = analogRead(A0);
  // print out the value you read:
  Serial.println(sensorValue);
  delay(1);        // delay in between reads for stability
}
```

GrumpyOldPizza, 2019-03-04:
Don't need to try that. Of course that works on Butterfly. The only thing is that you could have selected a clock setting in the IDE that does not allow USB, or you have USB disabled.

larjohn, 2019-03-04:
What clock setting works for you?

GrumpyOldPizza, 2019-03-04:
The IDE tells you if USB gets disabled. Everything above or equal to 24MHz can do USB.

larjohn, 2019-03-04:
Unfortunately, it doesn't work. To be honest, even the manufacturer's example only starts the port the first time it is connected. If I reconnect the board to USB it cannot be started (code 10) or it is busy so I cannot monitor it. 

With arduino-STML32L4 it does not show at all.

From the datasheet, it seems it doesn't have notable differences compared to Butterfly.

GrumpyOldPizza, 2019-03-04:
Ok, does it work with a Butterfly board, yes or no ?

Did you connect on your custom design PB2 to USB_VBUS, yes or no ?

larjohn, 2019-03-04:
Sorry, I don't have a Butterfly to test. I only have the Radino L4 from In-Circuit and hoped your implementation would also work for me, as In-Circuit conveniently hide that they haven't implemented I2C, although they advertise it supports it. Your I2C works fine, and I would like to buy you a pizza (pun intended) for that!

Now only Serial communication is left for the board to be fully functional and go on with UWB Ranging, RPi, Scratch and the rest!

So, I tried my board in a different computer, and had the following results:

1. Using the manufacturer's library the COM port works (I can even connect with PuTTY)
2. Using arduino-STM32L4 no COM port.

Their core includes a folder with a USB_Device/CDC_Standalone library from MCD Application Team copyrighted by STMicroelectronics. There is also a matching middleware. All this code is available here: http://library.radino.cc/Arduino_1_8/radinoL4/radinoL4-0.9.2.tar.bz2.

Is it possible that they have made some custom circuitry requiring custom code to work?

larjohn, 2019-03-04:
The variant I am using is here: http://wiki.in-circuit.de/images/e/e1/305000110A_radinoL4_DW1000.pdf and I have bought the breakout board and in the datasheet I can't find PB2

GrumpyOldPizza, 2019-03-05:
The Butterfly coded uses PB2 to detect whether USB is connected or not ... So if there is not PB2 and you are using the Butterfly variant, then it will not detect USB.

larjohn, 2019-03-05:
Can you point me to where you do that check? I could circumvent it for my case (USB is always connected) until I find a better way to do it.

larjohn, 2019-03-05:
Must be here:

https://github.com/GrumpyOldPizza/arduino-STM32L4/blob/bacc184288a4644b2ee6a97672334983f3e788ab/cores/stm32l4/USBCore.cpp#L65

but what would be the always enable value? 1?

larjohn, 2019-03-05:
Then you go here: 

 ```
 if (stm32l4_gpio_pin_read(usbd_pin_vbus))
	    {
		usbd_pin_vbus_count = 10;
		
		armv7m_timer_start(&USBD_VBUSTimer, 1);
	    }
	    else
	    {
		usbd_pin_vbus_count = 0;

		armv7m_timer_start(&USBD_VBUSTimer, 50);
	    }
```
Can you explain what happens in these cases?

GrumpyOldPizza, 2019-03-05:
variant.h ... GPIO_PIN_NONE

larjohn, 2019-03-05:
Even better. Going to test that ASAP

larjohn, 2019-03-05:
Well it works! Your project has saved me of hundreds of hours understanding how to develop I2C support, let alone your support here, for an unsupported board.

I would like to reward this time spent. Do you accept donations of any kind?

kriswiner, 2019-03-05:
Hi Lazaros,

In addition to any monetary reward you might want to offer Thomas for his
fantastic Arduino core, you might buy some products from our Tindie stores
(Pesky Products and Tlera Corporation), and, of course, tell your friends
what great software and hardware we provide!

Kris

larjohn, 2019-03-05:
Excellent products in good prices...but I am based in EU. I hope, someday, it won't be like 2xtimes the price to buy something from the US (shipping + taxes + import fees may get bigger than the actual price)

kriswiner, 2019-03-05:
Order in quantity, and I will set a low price on the customs form. Shipping
costs what it costs, nothing I can do about that...

closed 2019-03-05 by larjohn