GrumpyOldPizza/arduino-STM32L4 #45: Ladybug MSC/DOSFS

flybrianfly, 2019-06-04:
I'd like to use USB/MSC with the Ladybug and a SD card on an SPI interface. I must be missing something fairly trivial since I'm not able to get this to work. Are there any examples for this functionality?

Thanks!
Brian

kriswiner, 2019-06-04:
Need a bit more information here, what are the connections and what does
"can't get it to work" mean?

flybrianfly, 2019-06-04:
Thanks for the quick reply. Connections are CLK to pin13, MOSI to pin 12, MISO to pin 11, and CS to pin 10. Power and ground from the Ladybug GND and VIN. I'm not able to get the SD to show up as a mass storage device on Windows.

kriswiner, 2019-06-04:
Did you select usb + mass srorage device and sd card in the tools dropdown
menu?

closed 2021-03-10 by flybrianfly