GrumpyOldPizza/arduino-STM32L4 #48: Butterfly analogWrite and pinMode conflicts

jacky4566, 2019-09-17:
There seems to be a problem with calling pinMode multiple times and using analogWrite.
The following code will fail for me:

> static const int MOSFET = 9;
> 
> void setup() {
>   //ON
>   pinMode(MOSFET, OUTPUT);
>   analogWrite(MOSFET, 255);
>   delay(1000);
>   //OFF
>   analogWrite(MOSFET, 0);
>   delay(3000);
>   //ON
>   pinMode(MOSFET, OUTPUT);//This pinMode causes problems
>   analogWrite(MOSFET, 255);
>   delay(1000);
>   //OFF
>   analogWrite(MOSFET, 0);
>   pinMode(MOSFET, INPUT);
> }
> 
> void loop() {
>   delay(1);
> }

Quite a few of the Arduino universe libraries use this format so its a bit frustrating to change them all.

GrumpyOldPizza, 2019-09-17:
What is the problem exactly ? Hang ? Wrong value ? Spike ?

jacky4566, 2019-09-17:
Ah sorry. The code continues but no output voltage is observed.

kriswiner, 2019-09-17:
Why are you calling this a second time? Not necessary.

jacky4566, 2019-09-17:
Yes I understand its not necessary but many Arduino libraries do this. This behavior is also very unexpected and could easily confuse new users. 
I found this while using an older [Motor Driver Library](https://www.arduinolibraries.info/libraries/motor-driver-library)

Its an easy fix to remove the pinMode() calls but still not great practice since pinMode() should be repeatable under Arduino. 

This also kills Charlieplexing LEDs where you need to change between INPUT and OUTPUT rapidly.

kriswiner, 2019-09-17:
Changing the mode should work fine, repeatedly calling the same mode is no
good practice.

jacky4566, 2019-09-17:
Changing the mode also does not work, see my example below. Repeatedly calling the same mode may not be good practice no, but it happens. [Even in Adafruit's libraries](Adafruit-Motor-Shield-library )

I feel this should be a concern for you guys since it's common enough in the Arduino enviroment.

```c++
static const int MOSFET = 9;

void setup() {
  //ON
  pinMode(MOSFET, OUTPUT);
  analogWrite(MOSFET, 255);
  delay(1000);
  //OFF
  analogWrite(MOSFET, 0);
  delay(3000);
  //ON
  pinMode(MOSFET, INPUT);
  pinMode(MOSFET, OUTPUT);
  analogWrite(MOSFET, 255); //No output produced. 0v
  delay(1000);
  //OFF
  analogWrite(MOSFET, 0);
  pinMode(MOSFET, INPUT);
}

void loop() {
  delay(1);
}
```

kriswiner, 2019-09-17:
Comment out these two lines and it should work:

pinMode(MOSFET, INPUT);
  pinMode(MOSFET, OUTPUT);

Why are you toggling the mode for no reason here?

If you insist on doing so, at least put a delay between the calls.

jacky4566, 2019-09-17:
Kris the code is just a simplified example of how to cause the issue.
A delay of 50ms also did not fix the problem. 

Your free to do as you wish but I believe that repeated calls to "pinMode(  , OUTPUT);" should not cause a failure of the pin to conduct.