GrumpyOldPizza/arduino-STM32L4 #49: Method to check for USB_VBUS

jacky4566, 2019-10-09:
Is there a method to check for USB_VBUS? 

I see it is tied to GPIO_PIN_PB2.
I tried to read pin 27 but it just returns false.

My intention is to shutdown the device when USB is not connected.

GrumpyOldPizza, 2019-10-09:
The USB device is shut down already as long as there is no USB_VBUS detected. So what you are asking for is already implemented.

jacky4566, 2019-10-09:
Right but how do I check for that in my program? Otherwise nothing gets printed because the device is sleeping. 

```C++

 void setup() {
   Serial.begin(9600);
 }
 
 void loop() {
   if (/*Check for USB*/) {
     Serial.println(STM32.getTemperature());
     delay(1000);
  } else {
     STM32.stop(1000);
   }
 }

```

kriswiner, 2019-10-09:
VBUS = STM32L0.getVBUS();

jacky4566, 2019-10-09:
We are talking about the L4 here. That function doesn't exist in STM32.h

jacky4566, 2019-10-09:
I found a workaround by changing the variant.cpp file for my board.
Changing Null to GPIOB allows me to read the pin. Is there any disadvantage to doing this?

https://github.com/GrumpyOldPizza/arduino-STM32L4/blob/ac659033eadd50cfe001ba1590a1362b2d87bb76/variants/STM32L433CC-Butterfly/variant.cpp#L87

kriswiner, 2019-10-15:
USBConnected = USBDevice.connected();

jacky4566, 2019-10-15:
Great that seems to work. This should be in the documentation.

```C++
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_BUILTIN, USBDevice.connected());
}
```

closed 2019-10-15 by jacky4566