GrumpyOldPizza/arduino-STM32L4 #50: DAC Speed under 1Khz

jacky4566, 2019-12-11:
Is there anything I can do to improve the speed of the DAC? 
Or is there a better way to update the DAC value?

Using this code i can only achieve 769hz when I would like to get up into audio ranges 15kHz.

```C++
void setup() {
}

void loop() {
  analogWrite(PIN_DAC0, i++);
  if (i >= 255) {
    i = 0;
  }
}
```

jacky4566, 2019-12-14:
[So doing some reading to get a faster DAC ](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=2&ved=2ahUKEwioitzS_7PmAhWJGDQIHSzrBogQFjABegQIARAC&url=https%3A%2F%2Fwww.st.com%2Fresource%2Fen%2Fapplication_note%2Fdm00129215.pdf&usg=AOvVaw0xUwhvkNRusSxtOw53aWTv) ST recommends using the DMA to move information to the DAC as its ready. 
I tried my hand to use your DMA library but I cant seem to get it to fire. My code will complile but nothing happens :/

DMA feeding DAC code:
```C++
#include "stm32l4_dma.h"

stm32l4_dma_t stm32l4_DAC_DMA;

static int DACresolution = 12;

uint32_t i;

uint32_t DAC_DMA_OPTION = DMA_OPTION_MEMORY_TO_PERIPHERAL |
                          DMA_OPTION_PERIPHERAL_DATA_SIZE_32 |
                          DMA_OPTION_MEMORY_DATA_SIZE_32 |
                          DMA_OPTION_MEMORY_DATA_INCREMENT |
                          DMA_OPTION_PRIORITY_MEDIUM ;

void setup() {
  Serial.begin(9600);
  while (!Serial) {}
  analogWriteResolution(DACresolution);
  analogWrite(PIN_DAC0, i);
  analogWrite(PIN_DAC1, i);

  DAC->CR = (uint32_t) DAC->CR | 0x3000; //enable DAC DMA and underrun

  stm32l4_dma_create(&stm32l4_DAC_DMA, DMA_CHANNEL_DMA1_CH3_DAC1, DMA_OPTION_PRIORITY_LOW);
  stm32l4_dma_enable(&stm32l4_DAC_DMA, NULL, NULL);
}

void loop() {
  i = (i + 64) & DAC_DHR12R1_DACC1DHR;

  stm32l4_dma_start(&stm32l4_DAC_DMA, (uint32_t)&DAC->DHR12R1,(uint32_t)i, 1, DAC_DMA_OPTION);    //This should trigger a DMA trandfer to the DAC->DHR12R1 register.
  
  if (i >= pow(2, DACresolution)) {
    i = 0;
  }
}
```

GrumpyOldPizza, 2019-12-14:
Almost there ;-)

First off, you should be using circular DMA instead of oneshot. Use the half-done / full-done to implement a simple double buffered scheme.

In DAC->CR, you need to enable an external trigger (no underrun or such). The external trigger (TSEL) should be set to TIM7_TRIG. TIM7 is used for Tone()/noTone(), so it seems appropriate to use that one.

You can look at the stm32l4_wiring_tone.c code to see how it's kind of used, except you don't need to interrupt callback.

jacky4566, 2019-12-14:
Thanks for the help GrumpyOldPizza.
Still a bit stuck here. I was trying one shot mode just to get things working but circular mode also doesn't seem to be helping.

I have implement the Timer7 at 1MHZ and enabled the appropriate TSEL and TEN1. The DAC appears to be working off that source now. Plugging values into the second channel (DHR12R2) transfer correctly. 

With a circular DMA how does it know when to transfers? Do i need to point the Timer 7 to the DMA as well?

Here is the code so far. 
```C+
#include "stm32l4_dma.h"
#include "stm32l4_timer.h"

stm32l4_dma_t stm32l4_DAC_DMA;
stm32l4_timer_t stm32l4_DAC_timer;

static int DACresolution = 12;

uint32_t desiredValue;

uint32_t DAC_DMA_OPTION = DMA_OPTION_MEMORY_TO_PERIPHERAL |
                          DMA_OPTION_CIRCULAR |
                          DMA_OPTION_PERIPHERAL_DATA_SIZE_32 |
                          DMA_OPTION_MEMORY_DATA_SIZE_32 |
                          DMA_OPTION_PRIORITY_MEDIUM ;

void setup() {
  Serial.begin(9600);
  analogWriteResolution(DACresolution);
  analogWrite(PIN_DAC0, 0); //use board manager to setup dac
  analogWrite(PIN_DAC1, 0); //use board manager to setup dac

  //Setup timer 7 to trigger arbitraty 1MHz (@80MHZ)
  stm32l4_timer_create(&stm32l4_DAC_timer, TIMER_INSTANCE_TIM7, NULL, 0);
  stm32l4_timer_enable(&stm32l4_DAC_timer, (stm32l4_timer_clock(&stm32l4_DAC_timer) / 4000000) - 1, 4 - 1, TIMER_OPTION_COUNT_PRELOAD, NULL, NULL, 0);
  
  //Setup DMA to transfer into DAC
  stm32l4_dma_create(&stm32l4_DAC_DMA, DMA_CHANNEL_DMA1_CH3_DAC1, DMA_OPTION_PRIORITY_MEDIUM);
  stm32l4_dma_enable(&stm32l4_DAC_DMA, NULL, NULL);
  stm32l4_dma_start(&stm32l4_DAC_DMA, (uint32_t)&DAC->DHR12R1, (uint32_t)desiredValue, 1, DAC_DMA_OPTION);

  //enable DAC DMA, Trigger on TIM7, Timer trigger EN
  DAC->CR = (uint32_t) DAC->CR | 0x1014;

  //start timer
  stm32l4_timer_start(&stm32l4_DAC_timer, false);
}

void loop() {
  desiredValue = desiredValue + 64;
  DAC->DHR12R2 = desiredValue; //Verify operation on channel 2
  delay(100); //keep things slow
}
```