GrumpyOldPizza/arduino-STM32L4 #51: PA15 state at reset? not input

hallard, 2020-03-11:
@GrumpyOldPizza,

I'm using a dragonfly like board designed by Kris and we used PA15 as output to enable a LDO via an enable pin. We have placed a pull down on PA15 so at reset state the device behind LDO should be disabled.

unfortunately looks like reset state of this pin is not input and thus enable the LDO. Reconfiguring the PA15 pin as input in setup() solved the issue but wondering why and where this pin could be setup elsewhere not to be an input?

Can we easily fix?

of course our variant does not define PA15 as VUSB.
````
#define STM32L4_CONFIG_USB_VBUS           GPIO_PIN_PA9
#define PIN_USB_VBUS        (27ul)
#define PIN_USB_DM          (28ul)
#define PIN_USB_DP          (29ul)

/ 27..29 - USB (VBUS,DM,DP)
{ NULL,  GPIO_PIN_MASK(GPIO_PIN_PA9),  GPIO_PIN_PA9,            0,                                                              PWM_INSTANCE_NONE,  PWM_CHANNEL_NONE, ADC_INPUT_NONE },
{ NULL,  GPIO_PIN_MASK(GPIO_PIN_PA11), GPIO_PIN_PA11,           0,                                                              PWM_INSTANCE_NONE,  PWM_CHANNEL_NONE, ADC_INPUT_NONE },
{ NULL,  GPIO_PIN_MASK(GPIO_PIN_PA12), GPIO_PIN_PA12,           0,                                                              PWM_INSTANCE_NONE,  PWM_CHANNEL_NONE, ADC_INPUT_NONE },

````

Thanks

GrumpyOldPizza, 2020-03-11:
PA15 per default comes up as JTDI. So that *should* have been a INPUT with a PULLUP. So I suppose a stronger external pulldown should work.

I suppose one could switch PA15 earlier in the boot process to be a INPUT, but I am not sure that is worth the extra trouble, if you later on in "setup()" change it anyway.

kriswiner, 2020-03-11:
We are using a 100 K pulldown, so maybe the solution to this "problem" is
simply to use a 10 K pulldown.

hallard, 2020-03-11:
Yep nice point pull up should be approx 50k and explain the default 2.4v reading with R divider 100K 
10K should do the job and I agree changing it in framework may hurts other board. Anyway I need to disable pull up on setup to avoid extra current so best is to leave as is.
Know we know and know how to fix. 
Would be interesting to know if this applies to some other  I/O also