GrumpyOldPizza/arduino-STM32L4 #54: Maximum Baud Rate

flybrianfly, 2023-09-14:
Am I correct, guessing from [this file](https://github.com/GrumpyOldPizza/arduino-STM32L4/blob/ac659033eadd50cfe001ba1590a1362b2d87bb76/system/STM32L4xx/Source/stm32l4_uart.c#L657) that the maximum supported baud rate is 921600?

kriswiner, 2023-09-14:
I think this is right. Not sure how much higher you might need to go....

GrumpyOldPizza, 2023-09-14:
921600 is the limit.

May I ask why this high transfer rate ?

flybrianfly, 2023-09-14:
Transferring data between a Ladybug and an ESP32. The Ladybug is doing the heavy lifting to determine the states of a vehicle and the ESP32 is wirelessly broadcasting those states to a display - don't have a hard requirement on baudrate, just trying to minimize latency. I can work on compressing the packet size, but if I could increase the baudrate, that either provides an easier solution to reducing latency or provides options to reduce it further.