GrumpyOldPizza/arduino-STM32L4 #55: stm32L433CCT6, will program through the USB, however, no debug...

joey120373, 2024-12-02:
I built a board based on the Butterfly, much of the schematic was borrowed from the butterfly schematics. I added a DCDC converter and a couple extra connections, and put it on a RPi2040 type board form factor. I am able to get a sketch to load just fine, blink for example, and everything works great, love the low power!

However, when i load a simple UART scetch, like the "ASCII TABLE" arduino example scetch, it loads fine, but no Serial port ever shows up. I have tried on both my widows machine and My MAC. 

The Butterfly board schematics were relatively hard to read when in regards to the USB connection, there appears to be a TVS diode, and a second component??, its this second component that illudes me, it appears to be passive, so im assuming its just a resistor set?  I borrowed the circuit from a DUE schem for the protection. a couple 22ohm resistors and a couple inductors, Im wondering if this could be causing the issue?

I did make sure that the USB data lines were the same length, and have a nice straight shot to the STM pins, but im no EE, so its possible there it may be just a bad design. But since it (they) programs fine ( all three boards i built, actually ) I am assuming that the USB is at least somewhat functional.

kriswiner, 2024-12-02:
Stipulated that the EAGLE CAD schematic for Butterfly is not great.

STM32 MCUs do not require 22R terminators on the USB data lines. I use a
HSP-061 TVS diode as recommended by ST but this is also not really required.

We employ a GPIO (PB2) as USB detect to shut down the USB controller when
USB is not detected. Did you connect this in your design?

joey120373, 2024-12-07:
Ahh, could be, I put a jumper from "VBus" ( USB +5v ) to pin PB2, I must have put that there because I didn't understand what that was for, so possibly all i need to do is close that jumper so that PB2 is high when USB is connected?
Thanks, ill give that a try.
Joe

kriswiner, 2024-12-07:
Maybe...at least this is one thing I know could cause your trouble...

joey120373, 2024-12-08:
Yep! that did it, I designed this board as a quick and dirty trial run to see if i could simply just swap controllers on an existing board. I wanted to play around little bit though, without getting too deep into the feature creep. 
I don't remember exactly why I added the jumper, but i think I was being cautious with jamming 5+v onto a controller pin ( even if it is a 5v compatable pin ). In any case, I added a configurable power supply, i can choose either a pico inspired buck/boost converter, or the LDO used on the butterfly. I settled on a pico form factor just for kicks, and I added a Adafruit "EYESPI" FFC connector to the end just cause why not. 
The project this will be added to, I desoldered the ATMEGA328 ( running at 8Mhz ) and wired up the STM32L433 board to the signals im using : SPI, IIC and a half doezen digital IO pins. I figured i was in for a lot of hair pulling getting the code ported over, but !!! I literally just changed the board type in arduino, hit send and waited for the laundry list of compiler errors to show up.......to my suprise, it compiled, loaded and the board just works! 10 times the clock speed, lower power and 32 bit goodness. You guys have done a great job on this, cant thank you enough. The board is a loop powered 4-20mA transmitter, it uses a I2C DAC to control the current, an I2C ADC to monitor the current, and a rotory encoder and an SPI LCD for user interface. The ATMEGA was fine, but just, and i was running out of room for the code. This helps on all fronts, it gives more headroom to keep the min current below 4mA, and i now have more room for code to add some much needed functionality.

kriswiner, 2024-12-08:
glad it worked for you!